IvLabs/autonomous-delivery-robot #1: Semantic Segmentation testing on Jetson TX1

akshaykvnit, 2019-10-13:
We need to test the semantic segmentation model (of Keras/TF) on Nvidia Jetson TX1, calculate the FPS and identify whether we require additional hardware for this computation.

akshaykvnit, 2019-10-15:
This task requires installation of TensorFlow on Jetson. Currently, there is some issue with the setup of Jetson (the OS is not on SD card), so enough space is not available for TF installation. Workaround is provided [here](https://www.jetsonhacks.com/2017/01/26/run-jetson-tx1-sd-card/).
Will add TF installation link later.

akshaykvnit, 2019-10-22:
The installation of Jetpack 3.1 has CUDA 8.0, while TensorFlow requires CUDA 9.0. So, we tried the installation of both Jetpack 3.3 and 3.2 which has CUDA 9.0, but there were some issues during the installation, so it needs to be repeated. This will be done after coming back from home (have to work on FYP presentation now).

akshaykvnit, 2019-10-24:
Take a look at testing videos processed on the GTX 1080Ti at [this link](https://drive.google.com/drive/folders/1ZWK42bZJJxiEu8831j1k6gu-Vfkt8kwf?usp=sharing).

akshaykvnit, 2019-12-08:
Since, older versions (16.04) weren't working, we tried using JetPack 4.2.1 (18.04). However, the shifting to SD card workaround mentioned earlier (from [here](https://www.jetsonhacks.com/2017/01/26/run-jetson-tx1-sd-card/)) doesn't work directly in 18.04. So, we'll try to install the older JetPack, then shift to SD card and then install PyTorch. Hopefully, this will work and we can shift our code to PyTorch. 
Worst case scenario, we'll have to use a Raspberry Pi 3.

akshaykvnit, 2019-12-10:
Shifted 18.04 Ubuntu to SD Card (using same link mentioned above). Installed PyTorch 1.3.0 using [this link](https://elinux.org/Jetson_Zoo#PyTorch_.28Caffe2.29) for Python 2 because ROS supports only Python2.

Will shift all code to PyTorch now. This will take at least one week.

akshaykvnit, 2019-12-13:
Shifted all code to PyTorch (Python2). Tested the code on Jetson TX1, got around 8 FPS for inputs of 640x320. However, network isn't trained fully yet. Need to do this, but the code works on Jetson, so this task is done.

navidpanchi, 2019-12-14:
You have predicted on a single image at a time, you  should use batch prediction. That could give many more fps than this.

akshaykvnit, 2019-12-15:
What do we do with those multiple predictions? I think we need only the latest one. Unless we are thinking of averaging the predictions of the last few images. 

Another problem is that Jetson has only 4 GB of VRAM, and predicting for single image takes 3.5 GB. I'll still try batch prediction and update this comment.

closed 2019-12-31 by akshaykvnit