IvLabs/autonomous-delivery-robot #4: Review of last year's related projects and thesis

akshaykvnit, 2019-10-13:
Contact Shubhanshu sir and discuss their hardware for autonomous RC car. Also request him to provide their thesis (for detailed review).

akshaykvnit, 2019-10-14:
Things to be discussed: 
- All major hardware details (he will know what parts were critical)
- Mounting BLDC motor and specs of BLDC motor
- Which servo motor was used? Original one or separate?
- Limitations in the hardware which they encountered
Add more (if any)

Rajeshree73, 2019-10-14:
Hardware Modifications:
for steering: Servo 
for wheels: BLDC 
Changed Ic engine to Battery
Air suspension to hydraulic suspension 
Problem Faced: Angle given to the servo wasn't actually controlled so fluctuation was there. also, the voltage wasn't regulated so speed was varying

akshaykvnit, 2019-10-14:
@Rajeshree73 Didn't get the part about hydraulic suspension. Could you elaborate? Also we need to discuss how to overcome the limitations. So, continue the conversation here itself on these points.

Rajeshree73, 2019-10-15:
For getting the voltage constant we can add a voltage regulator circuit and to get the constant angle we can design the feedback loop control system

Rajeshree73, 2019-10-15:
Link for the Final year thesis(Shubhanshu):
https://drive.google.com/file/d/1Ux5wBNMtnJSvihENM5TFg6IVGhjMgxr0/view?usp=sharing

closed 2019-12-10 by akshaykvnit