IvLabs/autonomous-delivery-robot #7: Combine 2D LIDAR, cameras, and depth map data

akshaykvnit, 2019-10-13:
Combine all vision sensors data to get a 360 degree depth map which can be used for local planning.

aj161198, 2019-10-20:
I have finished stitching images without feature matching. This is done for two cameras as my laptop can support at max 2 cameras. Also calibration of all lab cameras is done as well as the algorithm for more than two cameras is designed but it is yet to be tested on a more powerful computer.

patilunmesh, 2020-03-08:
I have combined contour of the segmented mask and lidar data, to get a 360 degree ring of obstacles

patilunmesh, 2020-03-08:
Can we combine depthimage and segmented mask so that the lidar data and image point data will work well. @akshaykvnit  @aj161198

closed 2021-11-05 by akshaykvnit