IvLabs/autonomous-delivery-robot #15: Interfacing of GPS module

akshaykvnit, 2020-01-17:
- We need to interface a GPS module to get the global coordinates of the robot. Accordingly, we bought [Ublox M8N GPS module](https://robu.in/product/ublox-neo-m8n-high-precision-gps-built-compass-apm-amp2-6-apm-2-8/?gclid=Cj0KCQiA04XxBRD5ARIsAGFygj-emF8q7k1wUKP767svl779wMm2BSoBQNHN1UG03ylP-fkdu70zmxEaAsIvEALw_wcB). 
- There is an extra JST connector on it (with 2 pins). Ignore that when interfacing with Arduino. However, that will be used if interfacing with Pixhawk FC or similar hardware which expects 6 pins instead of 4 for GPS.
- Also, remember that most GPS modules work on 3.3V, so we can't connect Rx and Tx pins of Arduino (which are 5V) directly. It might work directly but safer to use a logic level converter (or simply a resistor divider circuit). Also, remember to use common ground for everything.
- Note that after powering on the GPS module, it takes around 30 seconds to 1 minute to get the GPS fix. So, allow that much time before labelling the module as broken.
- Tried using [this library](https://github.com/SlashDevin/NeoGPS). It has good instructions. The basic code given in this produces UTC time, date and latitude, longitude, altitude, speed information as a demo. However, when using outdoors (on the road outside the Academic Block), it doesn't work correctly. **It gives the UTC time and date correctly but doesn't give any position information.** The location used for testing is pretty much open to the sky and time of testing was around 1:15 AM IST.
- Need to test in more places. Not done immediately since it's 1:30 AM and don't want to roam around campus alone at night (lot of wild animals around :grimacing: :joy:). Will be done in the morning/afternoon (today itself).

patilunmesh, 2020-01-25:
$GNGLL,2107.41517,N,07903.00047,E,101635.00,A,D*7E
$GNTXT,01,01,01,More than 100 frame errors, UART RX was disabled*70
$GNRMC,101636.00,A,2107.41516,N,07903.00039,E,0.029,,250120,,,D*63
$GNVTG,,T,,M,0.029,N,0.053,K,D*35
$GNGGA,101636.00,2107.41516,N,07903.00039,E,2,11,1.57,334.6,M,-63.7,M,,0000*68
$GNGSA,A,3,23,04,08,11,27,01,41,09,,,,,2.19,1.57,1.53*17
$GNGSA,A,3,84,73,85,,,,,,,,,,2.19,1.57,1.53*17
$GPGSV,4,1,15,01,10,147,32,03,04,177,21,04,48,155,42,07,44,341,*70
$GPGSV,4,2,15,08,54,062,45,09,77,208,17,11,35,130,40,17,01,205,*72
$GPGSV,4,3,15,18,,,34,23,54,163,42,27,23,042,41,28,25,253,*48
$GPGSV,4,4,15,30,21,312,,40,53,231,,41,65,169,38*4C
$GLGSV,3,1,11,65,17,320,,70,01,172,,71,38,206,,72,46,280,*65
$GLGSV,3,2,11,73,26,036,31,74,33,339,,75,13,292,,83,16,044,21*6F
$GLGSV,3,3,11,84,38,105,35,85,17,165,32,,,,40*6C

getting this message in a loop. blue light is blinking indicating that fix is received but unable to extract it maybe.

patilunmesh, 2020-01-25:
also sometimes this output:

$GNRMC,103339.00,A,2107.41402,N,07903.00107,E,0.219,,250120,,,D*62
$GNVTG,,T,,M,0.219,N,0.405,K,D*33
$GNGGA,103339.00,2107.41402,N,07903.00107,E,2,11,1.65,332.8,M,-63.7,M,,0000*61
$GNGSA,A,3,23,04,08,11,27,01,41,09,,,,,2.40,1.65,1.75*1E
$GNGSA,A,3,84,73,85,,,,,,,,,,2.40,1.65,1.75*1E
$GPGSV,4,1,15,01,14,141,35,03,00,172,,04,40,158,41,07,48,349,*75
$GPGSV,4,2,15,08,49,052,46,09,70,194,20,11,39,122,42,17,07,205,*70
$GPGSV,4,3,15,18,,,34,23,46,164,43,27,17,039,39,28,29,261,*44
$GPGSV,4,4,15,30,25,318,,40,53,231,,41,65,169,37*4D
$GLGSV,3,1,09,65,22,313,,71,29,201,,72,44,265,,73,24,046,36*63
$GLGSV,3,2,09,74,34,349,,75,15,301,,83,12,036,,84,42,093,34*69
$GLGSV,3,3,09,85,25,161,35*56
$GNGLL,2107.41402,N,07903.00107,E,103339.00,A,D*75
$GNRMC,103340.00,A,2107.41400,N,07903.00102,E,0.135,,250120,,,D*66
$GNVTG,,T,,M,0.135,N,0.250,K,D*38
$GNGGA,103340.00,2107.41400,N,07903.00102,E,2,11,1.65,332.7,M,-63.7,M,,0000*67
$GNGSA,A,3,23,04,08,11,27,01,41,09,,,,,2.40,1.65,1.75*1E
$GNGSA,A,3,84,73,85,,,,,,,,,,2.40,1.65,1.75*1E
$GPGSV,4,1,15,01,14,141,35,03,00,172,,04,40,158,41,07,48,350,*7D
$GPGSV,4,2,15,08,49,052,46,09,70,194,21,11,39,122,42,17,07,205,*71
$GPGSV,4,3,15,18,,,34,23,46,164,43,27,17,039,39,28,29,261,*44
$GPGSV,4,4,15,30,25,318,,40,53,231,,41,65,169,37*4D
$GLGSV,3,1,09,65,22,313,,71,29,201,,72,44,265,,73,24,046,36*63
$GLGSV,3,2,09,74,34,349,,75,15,301,,83,12,036,,84,42,093,33*6E
$GLGSV,3,3,09,85,25,161,35*56
$GNGLL,2107.41400,N,07903.00102,E,103340.00,A,D*7C
$GNRMC,103341.00,A,2107.41394,N,07903.00108,E,0.192,,250120,,,D*6A
$GNVTG,,T,,M,0.192,N,0.355,K,D*31
and so on...............

no repeating messages

closed 2021-11-05 by akshaykvnit