IvLabs/autonomous-delivery-robot #16: Issue in powering Jetson TX1 using LiPO battery

Aniket-Gujarathi, 2020-01-22:
Tried to power the Jetson TX1 using 3S LiPO battery with output voltage of around 11V, but the jetson is not able to boot. 
1. According to [link](https://devtalk.nvidia.com/default/topic/1050443/jetson-tx2/critical-current-limit-jetson-tx2/) we tried to change the value of critical_current_limit0 by:
>>  echo 4000 > critical_current_limit0 
However, after rebooting jetson, the values revert to the initial default value.
2.  According to some [forums](https://devtalk.nvidia.com/default/topic/914529/battery-for-jetson-tx1/?offset=57) we attached a 4000uF capacitor across the input. But, even this didn't work out.

patilunmesh, 2020-01-22:
I have gone through some forums and also have talked with few people. I come to a conclusion that we should boost the 3s supply to 19v and then provide to Jetson. Also we need to check current on dc supply.

YogeshPhalak, 2020-01-27:
Attaching battery via a switch and other circuitry were limiting the current, which was causing the problem in booting Jetson. I have attached the DC power jack of jetson directly to the 3S LiPo battery, which solved the problem.

closed 2020-01-27 by YogeshPhalak

akshaykvnit, 2020-01-28:
> I have attached the DC power jack of jetson directly to the 3S LiPo battery, which solved the problem.

@YogeshPhalak Is that safe? Also, I think we need some capacitors to hold up the terminal voltage in case of brownouts (as Kamat sir had suggested earlier).